randdusing/cordova-plugin-bluetoothle #632: Installing Version 5.x.x not possible

ebeling, 2020-08-03:
Many thanks  for new release! How can I install this? I use current cordova-cli (9.0.0 with  cordova-lib@9.0.1). If I call 

`cordova plugin add cordova-plugin-bluetoothle`

following message appear:

`Unmet project requirements for latest version of cordova-plugin-bluetoothle:
    cordova (9.0.1 in project, >=100 required)
Fetching highest version of cordova-plugin-bluetoothle that this project supports: 4.5.14 (latest is 5.0.0)`

randdusing, 2020-08-06:
This is odd. I haven't changed any dependencies in a long time... I wonder if anyone else is having this problem.

randdusing, 2020-08-06:
Ahh, it must be related to this line:
https://github.com/randdusing/cordova-plugin-bluetoothle/blob/master/package.json#L35

Let me see what this actually does

ebeling, 2020-08-11:
Thank you for release 5.0.2. I have tested with cordova-cli 10.0.0 and it works now!

closed 2020-08-11 by ebeling