randdusing/cordova-plugin-bluetoothle #633: Problem with connect phones Android to Android

J4si3k, 2020-08-24:
I have prepared a basic setup according to the documentation and unfortunately I can't connect via the `bluetoothle.connect` method. I set one telephone as a central and the other as a peripheral. I execute the following instruction:

- Peripheral: initialize -> initializePeripheral -> addService -> startAdvertising (wait for central scan) -> stopAdvertising
- Central: initialize -> startScan (found phone on advertising) -> stopScan -> connect (by getting address like 76:56:59:05:34:55). I waiting for callback but without any message from success or failure. Also isConnected return `{"isConnected": false ...}`

I Also found that on run `addService` method on peripheral is throwing following error (from characteristics config in my opinion):
```
Attempt to invoke virtual method 'long java.util.UUID.getMostSignificantBits()' on a null object reference
```

What i do wrong? This process is correct? How to make a call between two phones?
Thanks in advance for any help.

randdusing, 2020-08-26:
It sounds like a UUID isn't being passed properly to addService()

J4si3k, 2020-08-26:
ok this problem with UUID went away but still can't connect to the other device. I have the following setup in Vue, maybe you will see what I am doing wrong.

```
export default {
    data() {
        return {
            central: {
                log: ''
            },
            logs: '',
            peripheral: {
                log: '',
                service: ''
            },
            name: '',
            devices: [],
            platform: null
        }
    },
    methods: {
        centralInit() {
            bluetoothle.initialize((res) => {
                this.central.log = JSON.stringify(res);
            }, {
                request: true,
                restoreKey : "My-some-app"
            });
        }, 
        successMessage(res) {
            this.log = `callback: ${JSON.stringify(res)}`;
        },
        failureMessage(res) {
            this.log= JSON.stringify(res)
        },
        startScanSuccess(result) {

            this.central.log = `startScan: ${result.status}`;

            if (result.status === "scanStarted") {
                this.central.log = "Scanning for devices..."
            }
            else if (result.status === "scanResult") {
        
                if (!this.devices.some(function (device) {
                    return device.address === result.address;
                })) {
                    this.central.log = `FOUND DEVICE: ${result.name}`;
                    this.devices.push(result);
                }
            }
        }, 
        centralScanStart() {
            bluetoothle.startScan(this.startScanSuccess, this.failureMessage, {
                "services": []
            });
        },
        centralConnect(address) {
            this.central.log = `connecting to: ${address}`;

            bluetoothle.connect((result) => {
                console.log(result.status)    
                this.central.log = `connected`;
            }, (e) => {
                console.log(`${JSON.stringify(e)}`)
                this.central.log = `not connected: ${JSON.stringify(e)}`;
            }, { address, autoConnect: true } );
        },
        peripheralInit() {
            bluetoothle.initializePeripheral(
                this.successPeripheralMessage, 
                this.failurePeripheralMessage, 
                {
                    "request": true,
                    "restoreKey": "My-some-app"
                }
            );
        },
        successPeripheralMessage(res) {
            this.peripheral.log = JSON.stringify(res);
        },
        failurePeripheralMessage(res) {
            this.peripheral.log = JSON.stringify(res)
        },
        create_UUID(){
            var dt = new Date().getTime();
            var uuid = 'xxxxxxxx-xxxx-4xxx-yxxx-xxxxxxxxxxxx'.replace(/[xy]/g, function(c) {
                var r = (dt + Math.random()*16)%16 | 0;
                dt = Math.floor(dt/16);
                return (c=='x' ? r :(r&0x3|0x8)).toString(16);
            });
            return uuid;
        },
        addSerivce() {
            // this.peripheral.service = this.create_UUID();
            this.peripheral.service = "1234";
            bluetoothle.addService(
                this.successPeripheralMessage, 
                this.failurePeripheralMessage, 
                {
                    service: this.peripheral.service,
                    characteristics: [
                        { 
                            uuid: this.create_UUID(),
                            permissions: {
                              read: true,
                              write: true,
                            },
                            properties : {
                              read: true,
                              writeWithoutResponse: true,
                              write: true,
                              notify: true,
                              indicate: true,
                            }
                          }
                    ]
            });
        },
        peripheralAdvertising() {
            this.addSerivce(); 
            bluetoothle.startAdvertising(
                this.successPeripheralMessage, 
                this.failurePeripheralMessage,
                {
                    "service": this.peripheral.service,
                }); 
        },
       //other methods...
}
```
Of course, I call all methods in the same order as I wrote above. I guess the bug must be somewhere in my setup

J4si3k, 2020-08-29:
I stopped using advertising, then the address stopped changing and I could connect.

closed 2020-08-29 by J4si3k