randdusing/cordova-plugin-bluetoothle #634: discover() with clearCache doesn't re-discover

pschlang, 2020-08-27:
Hi,

when calling discover() a second time with clearCache set to true, no re-discovery is performed, since the device is already in "discovery completed" state. With clearCache set to true, I would expect discover() to perform a re-discovery.

Patrick

pschlang, 2020-08-27:
If you agree, this patch would change the behaviour:

```
diff --git a/src/android/BluetoothLePlugin.java b/src/android/BluetoothLePlugin.java
index 3301c12..c217fb1 100644
--- a/src/android/BluetoothLePlugin.java
+++ b/src/android/BluetoothLePlugin.java
@@ -1662,18 +1662,20 @@ public class BluetoothLePlugin extends CordovaPlugin {

     addDevice(returnObj, device);

-    int discoveredState = Integer.valueOf(connection.get(keyDiscoveredState).toString());
-    //Already initiated discovery
-    if (discoveredState == STATE_DISCOVERING) {
-      addProperty(returnObj, keyError, errorDiscover);
-      addProperty(returnObj, keyMessage, logAlreadyDiscovering);
-      callbackContext.error(returnObj);
-      return;
-    } else if (discoveredState == STATE_DISCOVERED) {
-      //Already discovered
-      returnObj = getDiscovery(bluetoothGatt);
-      callbackContext.success(returnObj);
-      return;
+    if (obj == null || !obj.optBoolean("clearCache", false)) {
+      int discoveredState = Integer.valueOf(connection.get(keyDiscoveredState).toString());
+      //Already initiated discovery
+      if (discoveredState == STATE_DISCOVERING) {
+        addProperty(returnObj, keyError, errorDiscover);
+        addProperty(returnObj, keyMessage, logAlreadyDiscovering);
+        callbackContext.error(returnObj);
+        return;
+      } else if (discoveredState == STATE_DISCOVERED) {
+        //Already discovered
+        returnObj = getDiscovery(bluetoothGatt);
+        callbackContext.success(returnObj);
+        return;
+      }
     }

     //Else undiscovered, so start discovery
```

randdusing, 2020-08-29:
Mind raising a PR? Looks like the change should be good, but a little difficult to see.

pschlang, 2020-09-14:
Thanks for your response. Created the PR.

PR #638 (2020-09-19): Force re-discovery when clearCache is set
Resolves #634
commit 2c08b727 (2020-09-19): Force re-discovery when clearCache is set

Resolves #634
closed 2020-09-19 by randdusing