randdusing/cordova-plugin-bluetoothle #645: retrieveConnected() returns empty even if is connected

luruke, 2020-10-01:
Hello, i'm having problem using `retrieveConnected`, which returns empty array even though i'm connected with a device.

Here's a sample scenario:
1) Scanning devices
2) Found device then connecting

After that:
- retrieveConnected() returns empty array
- But if I use isConnected() with the device address, it does returns that is connected
- On new scanning, the connecting device doesn't appear, is it the expected behaviour?

randdusing, 2020-10-10:
retrieveConnected() is poorly named due to the name of the underlying iOS call. It only returns paired devices.

luruke, 2020-10-11:
Ok thanks @randdusing , that makes sense.

So, I guess my app should be responsible to keep the connected device, as there is no method to know to which devices I am currently connected, right?

randdusing, 2020-10-11:
Yah, usually your app would keep track of it itself

closed 2020-10-12 by luruke