randdusing/cordova-plugin-bluetoothle #649: Services for Android Scan Result?

chrisjdev, 2020-11-10:
I noticed Android doesn't offer discovered service UUIDs in the scan result, but it appears it could. Have you considered making a change like: https://github.com/chrisjdev/cordova-plugin-bluetoothle/commit/41c198497383ce3365d320946aa4658f37af0567 ? I was thinking I could use the service UUIDs to only display devices that my application could plausibly work with to the user (device name can be changed to anything) Thank you for the great plugin and for looking at this question.

aroncal, 2020-11-11:
You can use the `services` param on `startScan` to filter devices only to those having the service UUID you indicate:

https://github.com/randdusing/cordova-plugin-bluetoothle#startscan

services = An array of service IDs to filter the scan or empty array / null. This parameter is not supported on Windows platform yet.

chrisjdev, 2020-11-11:
This is true, I should've also said that my application is going to support 2 different kinds of devices (likely more in the future), so I need a way to also tell the specific service a given device supports, so I know how to work with it. I have considered waiting until the user decides to connect, and then using the result of discover to figure out how to talk to it. That would work, but I figure if Android works like iOS, then I would know how to talk to the device at scan time, rather than at connection time.

chrisjdev, 2020-11-18:
As it ends up, one of the devices I'm trying to discover doesn't often come with services, anyway, so it's rarely discovered on iOS if I provide services on startScan. I'll close this issue, thanks again.

closed 2020-11-18 by chrisjdev