randdusing/cordova-plugin-bluetoothle #650: Permission problem on android sdk 29

DominikTrenz, 2020-11-17:
The plugin does not work with android target sdk 29 by default. It just doesn't find any devices without any error message. If i manually grant every location permission and background location permissions to the app it works again. The problem is that google play enforces a build target of sdk 29 or newer for new releases, so i can't release new versions until this bug is fixed.

This breaks the app:

<preference name="android-targetSdkVersion" value="29" />

bramk, 2020-11-18:
@DominikTrenz are you sure you have the latest plugin version? I ran into your issue with the same problem, but  27bf1a5e5a275b61c6f0f5d4d89965efd292cf84 solved it for me.

randdusing, 2020-11-18:
If that latest version doesn't work, feel free to raise a PR. I'm limited to iOS devices now, so I can't personally test anything.

DominikTrenz, 2020-11-18:
I was using the latest npm version (6.1.0) and i also checked the resulting AndroidManifest for the "ACCESS_FINE_LOCATION" -which was there. But that doesn't work on sdk 29. On first install the App asks for "Allow location access while using the app" or "Deny" and if you choose to allow it it doesn't work. When you manually go into the permissions and enable "Always allow location access und background location access" then the library works again. I also tested multiple Android Devices and Versions - same results. This library seems to not work with sdk 29 as it is. If i locate the problem i will raise a PR