randdusing/cordova-plugin-bluetoothle #651: Sometimes `discover` just hangs up

luruke, 2020-11-25:
Sometimes, calling `discover` on Android just hangs up, without resolving.
Without it, it's impossible to write or read...

It's seems completely random, about 30% of the time.

Any idea what can be causing this?

foofel, 2020-11-26:
Same thing just happend to me but its not really random, it happens every time, no callback returns. Iam using the newest version on a Pixel 4a with Android 11. I got a stacktrace, maybe it helps:
`2020-11-26 02:58:05.978 19638-19771/de.kraftmaschine.cordova D/BluetoothGatt: discoverServices() - device: 24:6F:.....
2020-11-26 02:58:05.978 19638-19771/de.kraftmaschine.cordova E/BluetoothGatt: android.os.DeadObjectException
        at android.os.BinderProxy.transactNative(Native Method)
        at android.os.BinderProxy.transact(BinderProxy.java:540)
        at android.bluetooth.IBluetoothGatt$Stub$Proxy.discoverServices(IBluetoothGatt.java:1956)
        at android.bluetooth.BluetoothGatt.discoverServices(BluetoothGatt.java:992)
        at com.randdusing.bluetoothle.BluetoothLePlugin.discoverAction(BluetoothLePlugin.java:1694)
        at com.randdusing.bluetoothle.BluetoothLePlugin.execute(BluetoothLePlugin.java:351)
        at org.apache.cordova.CordovaPlugin.execute(CordovaPlugin.java:98)
        at org.apache.cordova.PluginManager.exec(PluginManager.java:132)
        at org.apache.cordova.CordovaBridge.jsExec(CordovaBridge.java:59)
        at org.apache.cordova.engine.SystemExposedJsApi.exec(SystemExposedJsApi.java:41)
        at android.os.MessageQueue.nativePollOnce(Native Method)
        at android.os.MessageQueue.next(MessageQueue.java:335)
        at android.os.Looper.loop(Looper.java:183)
        at android.os.HandlerThread.run(HandlerThread.java:67)`
or
`2020-11-26 02:59:46.804 19638-19771/de.kraftmaschine.cordova E/BluetoothGatt: android.os.DeadObjectException
        at android.os.BinderProxy.transactNative(Native Method)
        at android.os.BinderProxy.transact(BinderProxy.java:540)
        at android.bluetooth.IBluetoothGatt$Stub$Proxy.refreshDevice(IBluetoothGatt.java:1936)
        at android.bluetooth.BluetoothGatt.refresh(BluetoothGatt.java:1428)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.randdusing.bluetoothle.BluetoothLePlugin.refreshDeviceCache(BluetoothLePlugin.java:1702)
        at com.randdusing.bluetoothle.BluetoothLePlugin.discoverAction(BluetoothLePlugin.java:1691)
        at com.randdusing.bluetoothle.BluetoothLePlugin.execute(BluetoothLePlugin.java:351)
        at org.apache.cordova.CordovaPlugin.execute(CordovaPlugin.java:98)
        at org.apache.cordova.PluginManager.exec(PluginManager.java:132)
        at org.apache.cordova.CordovaBridge.jsExec(CordovaBridge.java:59)
        at org.apache.cordova.engine.SystemExposedJsApi.exec(SystemExposedJsApi.java:41)
        at android.os.MessageQueue.nativePollOnce(Native Method)
        at android.os.MessageQueue.next(MessageQueue.java:335)
        at android.os.Looper.loop(Looper.java:183)
        at android.os.HandlerThread.run(HandlerThread.java:67)`

foofel, 2020-11-27:
I have a solution that seems to work for me atm. The error happens if i try to (re)discover on an already connected connection. if i close the connection and open it seems to work.

MojioMS, 2021-06-08:
I have no success/error callbacks on android 8 every time. Never yet on 9 or 10.