randdusing/cordova-plugin-bluetoothle #653: Android: Peripheral advertisment not working

maschopp, 2020-12-02:
Hi folks.

I try to build an ionic app (capacitor) that should work as central and peripheral.

Everything works fine so far except the advertisment.

When I start advertising, scanners can't find my peripheral. 

Interestingly when my app is running in background and I start a pheripheral simulator app from the store (BLE Peripheral Simulator) scanners can then find my created service, connect and read the value of my characteristic. 
So the service with its characteristic is created successfully. Only advertisement is not working.

This is my code so far (_home.component.ts_):
```
import { Component, NgZone } from "@angular/core";
import { AndroidPermissions } from "@ionic-native/android-permissions/ngx";
import { BluetoothLE } from "@ionic-native/bluetooth-le/ngx";
import { Platform } from "@ionic/angular";

/* Is needed because the defined types for RespondParams do not include address which leads to an error when sending the response on Android */
declare module "@ionic-native/bluetooth-le/ngx" {
  interface RespondParams {
    address: any;
    /** This integer value will be incremented every read/writeRequested */
    requestId: number;
    /** base64 string */
    value: string;
    /** not documented */
    offset?: number;
  }
}

@Component({
  selector: "app-home",
  templateUrl: "home.page.html",
  styleUrls: ["home.page.scss"],
})
export class HomePage {
  public lastScan = new Date();
  public devices = [];

  constructor(
    public bluetoothle: BluetoothLE,
    private ngZone: NgZone,
    public plt: Platform,
    private androidPermissions: AndroidPermissions
  ) {
    this.plt.ready().then((readySource) => {
      console.log("Platform ready from", readySource);

      this.androidPermissions
        .checkPermission(
          this.androidPermissions.PERMISSION.ACCESS_FINE_LOCATION
        )
        .then(
          (result) => this.initBLE(result),
          (err) =>
            this.androidPermissions.requestPermission(
              this.androidPermissions.PERMISSION.ACCESS_FINE_LOCATION
            )
        );
    });
  }

  initBLE(result) {
    console.log(result);

    if (!result.hasPermission) {
      this.androidPermissions
        .requestPermission(
          this.androidPermissions.PERMISSION.ACCESS_FINE_LOCATION
        )
        .then((resp) => this.initBLE(resp));
    } else {
      this.bluetoothle
        .initialize({
          request: true,
          statusReceiver: false,
          restoreKey: "myapp",
        })
        .subscribe(
          (ble) => {
            console.log("ble", ble.status); // logs 'enabled'
            this.bluetoothle
              .initializePeripheral({
                request: true,
                restoreKey: "myapp",
              })
              .subscribe(
                (status) => this.peripheralSuccess(status),
                (err) => {
                  console.log(err);
                }
              );

            setInterval(() => {
              this.scan();
            }, 30000);
          },
          (err) => {
            console.log(err);
          }
        );
    }
  }

  scan() {
    this.devices = [];
    this.bluetoothle
      .startScan({
        services: [],
      })
      .subscribe(
        (device) => {
          if (device.status == "scanResult") {
            if (
              this.devices.filter((dev) => dev.address == device.address)
                .length == 0
            ) {
              this.ngZone.run(() => {
                this.devices.push(device);
              });
            }
          }
        },
        (err) => {
          console.log(err);
        }
      );
    setTimeout(() => {
      this.bluetoothle.stopScan();
      console.log(this.devices);
    }, 5000);
  }

  peripheralSuccess(status) {
    if (status.status == "enabled") {
      this.bluetoothle
        .addService({
          service: "5F43E28E-9067-4BDD-BD1F-6A20F93432F1",
          characteristics: [
            {
              uuid: "1A94A38F-64A3-45D8-8ADD-D673F69EA287",
              permissions: {
                read: true,
                write: false,
              },
              properties: {
                read: true,
                write: false,
              },
            },
          ],
        })
        .then((data) => {
          this.bluetoothle
            .startAdvertising({
              services: ["5F43E28E-9067-4BDD-BD1F-6A20F93432F1"],
              service: "5F43E28E-9067-4BDD-BD1F-6A20F93432F1",
              name: "SICK",
              includeDeviceName: false,
              mode: "balanced",
              connectable: true,
              timeout: 0,
              txPowerLevel: "high",
            })
            .then((success) => {
              console.log('[startAdvertising]' + JSON.stringify(success));
            })
            .catch((err) => {
              console.log(err);
            });
        })
        .catch((err) => {
          console.log(err);
        });
    }

    if (status.status == "readRequested") {
      if (
        status.service == "5F43E28E-9067-4BDD-BD1F-6A20F93432F1" &&
        status.characteristic == "1A94A38F-64A3-45D8-8ADD-D673F69EA287"
      ) {
        this.bluetoothle
          .respond({
            address: status.address,
            requestId: status.requestId,
            value: this.bluetoothle.bytesToEncodedString(
              this.bluetoothle.stringToBytes(`I'am sick!`)
            ),
          })
          .then((data) => {
            console.log(data);
          })
          .catch((err) => {
            console.log(err);
          });
      }
    }
  }
}
```

For testing I use a Samsung Galaxy S7 (Android 8) and a Samsung Galaxy S10+ (Android 10).

Any help would be appreciated.

maschopp, 2020-12-04:
Finally got it working.
If anyone else faces a similar problem, here is my solution:

I had to change the advertising and the service settings.
For some reason it started working after I set _includeDeviceName_ to true and set the _service uuid_ to a 4 char long string.

Here is what I ended up with:

_Params for addService_:
```
{
   service: "1337",
   characteristics: [
      {
           uuid: "1A94A38F-64A3-45D8-8ADD-D673F69EA287",
           permissions: {
                read: true,
                write: false,
           },
           properties: {
                read: true,
                write: false,
            },
        },
   ],
}
```

_Params for startAdvertising:_
```
{
   services: ["1337"],
   service: "1337",
   name: "SICK",
   includeDeviceName: true,
   mode: "lowPower",
   connectable: true,
   timeout: 0,
   txPowerLevel: "high",
}
```

closed 2020-12-04 by maschopp

TawsTm, 2021-12-15:
This results because the Service must be a 16-Bit UUID in Hexadezimal => 4-Bits for every character => 16/4 = 4 Characters from 0 to F

marcui13, 2023-02-03:
Hi, how are you?
I am trying to develop a similar application, the difference is that once I select a device to connect to, I must establish a text chat communication between both devices. 
Is this possible??