randdusing/cordova-plugin-bluetoothle #655: Subscribe can read only 20 bytes from characteristic

Patric, 2020-12-16:
Hi, when using subscribe I only get 20 bytes from the characteristic. With nRF Connect I get full string from the characteristic. I can write everything properly. Just reading with subscribe is limited to 20 bytes. When using read() I can "unable to read on return" promise rejected error. I'm testing this on Android, with Arduino Nano 33 IoT.

`
  getResponse(characteristicUID: string){
    this.bluetoothle.subscribe({
      "address": this.connectionInfo$.getValue().address,
      "service": this.PRIMARY_SERVICE_UID,
      "characteristic": characteristicUID//,
      //"response": "No"
    }).subscribe((operationResult) => 
    {
      
      let value = operationResult.value;
    
      console.log("New message received", JSON.stringify(operationResult));
      if(value != undefined){
        console.log("Received message: ", atob(value));
      }
      
    });`

adelin-mcbsoft, 2021-01-15:
I confirm it happens on Android, however on iOS works as expected ( -> meaning you can get the entire length of the notified string).

adelin-mcbsoft, 2021-01-15:
For iOS works as with more than 20 bytes, however on Android, I believe this is standard behavior.

I had an eye on [this thread](https://stackoverflow.com/questions/24135682/android-sending-data-20-bytes-by-ble), where most of the users suggest increasing MTU size _(which you can do starting with Android Lolipop)_, so using the [bluetoothle.mtu](https://github.com/randdusing/cordova-plugin-bluetoothle#mtu) method you are able to do so and  afterwards you can receive more than 20 bytes.

Not really a bug.