randdusing/cordova-plugin-bluetoothle #660: retrieveConnected returns empty array

czzplnm, 2020-12-23:
Using a Samsung S8 with Ionic/VueX/Capacitor, here is my code:

```
getConnected: ({commit, state, dispatch}) => {
    return new Promise((resolve, reject) => {
      window.bluetoothle.retrieveConnected(
        function (success) {
          if (success.length === 0) {
            console.log('retrieveConneccted is empty!');
          } else {
            commit('setIsConnectedTo', success)
          }
          console.log('getConnected: retrieveConnected: success:', success);

          resolve();
        },
        function (error) {
          commit('setIsConnectedTo', null)
          console.log('getConnected: retrieveConnected: error:', error);
          reject();
        },
        {
          "services": [
            SERVICE_180D,
            SERVICE_180F
          ]
        }
      );
    })
  },
```

retrieveConnected connected is always an empty array?

czzplnm, 2020-12-28:
So it seems this only returns the paired devices. How are you supposed to figure out what you are connected to then!?!?

MonksterFX, 2021-01-23:
Have an similar problem on Iphone. After hot reload my iphone stays connected to my device but there is no possibility to kill that connection.