randdusing/cordova-plugin-bluetoothle #663: 2 cordova compile errors

Jerez69, 2021-01-17:
Hi everybody,
I'm trying to build a cordova app based on angular and the cordova-plugin-bluetoothle.

cordova run android --device
-----------------------------------------------
....
:compileDebugJavaWithJavac - is not incremental (e.g. outputs have changed, no previous execution, etc.).
K:\Development\Projects\WormholeWar\Playground\Bluetooth\BlutoothLE\dist\mobile\platforms\android\src\com\randdusing\bluetoothle\BluetoothLePlugin.java:2808: error: local variable address is accessed from within inner class; needs to be declared final
            if(bluetoothDevice.getAddress().equalsIgnoreCase(address)){
                                                             ^
K:\Development\Projects\WormholeWar\Playground\Bluetooth\BlutoothLE\dist\mobile\platforms\android\src\com\randdusing\bluetoothle\BluetoothLePlugin.java:2811: error: local variable pin is accessed from within inner class; needs to be declared final
                bluetoothDevice.setPin(pin.getBytes());
                                       ^
Note: Some input files use or override a deprecated API.
Note: Recompile with -Xlint:deprecation for details.
2 errors
:compileDebugJavaWithJavac FAILED

BUILD FAILED
...
-----------------------------------------------

My environment:
**cordova -v**
10.0.0

**gradle -v**

------------------------------------------------------------
Gradle 6.2.2
------------------------------------------------------------

Build time:   2020-03-04 08:49:31 UTC
Revision:     7d0bf6dcb46c143bcc3b7a0fa40a8e5ca28e5856

Kotlin:       1.3.61
Groovy:       2.5.8
Ant:          Apache Ant(TM) version 1.10.7 compiled on September 1 2019
JVM:          1.8.0-25 (Microsoft 25.71-b00)
OS:           Windows 10 10.0 amd64

Does anyone have the same problem?
And does anybody have an idea what I could do?
(But not changing the jave code, I'm just able to deal with html5/typescript/C#/C++)

Thanks a lot
Bye
Marcus

alexhisen, 2021-07-09:
@Jerez69 did you ever figure this out?

This happens to me as well with older cordova-android (5.x).

Jerez69, 2021-07-10:
Hi Alex, I have sometimes Problems compiling cordova.Can you give me a bit more Information which Problems when you are faced? ByeMarcus Von: Alex HisenGesendet: Freitag, 9. Juli 2021 21:38An: randdusing/cordova-plugin-bluetoothleCc: Jerez69; MentionBetreff: Re: [randdusing/cordova-plugin-bluetoothle] 2 cordova compile errors (#663) @Jerez69 did you ever figure this out?This happens to me as well with older cordova-android (5.x).