randdusing/cordova-plugin-bluetoothle #665: Play Store policy compliance

NunoSav, 2021-01-22:
My app is currently being rejected by Play Store due to ACCESS_BACKGROUND_LOCATION.
I've removed this permission from the app but they still complain about it.

Has anyone gone through this and managed to comply with the policy without submitting the declaration for a feature that requires background access?

Thanks!

TomBrouws, 2021-01-22:
We're running into the same issue. I don't think I can explain to Google why we need background location permission to have Bluetooth functionality. Is there a particular reason why the plugin asks this permission? Can we get the option to remove it?

NunoSav, 2021-01-22:
I've filled an appeal and in there response they said:

> - Since your app does not need background location, please request to remove background usage and reach compliance:
>
>    - For any APKs targeting Android 10 or newer (SDK level 29 or higher):
>      - Remove the ACCESS_BACKGROUND_LOCATION permission from your app APK or app bundle
>      - If you’re using ACCESS_COARSE_LOCATION or ACCESS_FINE_LOCATION, examine your code paths and restrict usage to foreground purposes only (learn more)
>      - You should no longer see the Location declaration listed under App Content
>    - For any APKs targeting Android 9 or older (SDK level 28 or lower):
>      - If you’re already using ACCESS_COARSE_LOCATION or ACCESS_FINE_LOCATION, examine your code paths and restrict usage to foreground purposes only (learn more)
>      - In your console declaration, select “No” to the question “Does your app access location in the background in APKs or app bundles targeting Android 9 or older?”

I've removed ACCESS_BACKGROUND_LOCATION permission only so far.

Not sure i can:
> examine your code paths and restrict usage to foreground purposes only

without a PR to this plugin.

Also, regarding:
>In your console declaration, select “No” to the question “Does your app access location in the background in APKs or app bundles targeting Android 9 or older?”

Any idea where specifically we can do this?

NunoSav, 2021-01-22:
The version `6.1.1` introduced `ACCESS_BACKGROUND_LOCATION ` ([see issue](https://github.com/randdusing/cordova-plugin-bluetoothle/issues/650)) and i believe this is unnecessary. I can scan devices just fine on Android 10 with latest patch (December) with `FINE` and `COARSE` permissions.

capitaineflam25, 2021-01-22:
Same difficulties validating my app update, i'm going to switch back to 6.1.0

randdusing, 2021-01-22:
For people saying background location permission isn't necessary, which SDK are you targeting? That seems to make a difference

NunoSav, 2021-01-22:
I use the following configuration:
```
minSdkVersion = 21
compileSdkVersion = 30
targetSdkVersion = 30
```

NunoSav, 2021-01-25:
I've managed to resolve the issue with Play Store while using v6.1.0 of this plugin.

In the next few days i will try publishing new internal releases using v6.1.1+ with the `ACCESS_BACKGROUND_LOCATION` permission and see if Play Store rejects it. I will update this thread when i get the results.

jholzer-cciq, 2021-02-11:
@randdusing where have you seen or heard that background location permission is necessary for BLE scanning on any version of the Android SDK? I have been unable to confirm that. Background location _should_ only be required if an app needs to scan for Bluetooth devices while running in the background. For apps that will only scan in the foreground, it sounds like `COARSE` and `FINE` permissions are sufficient.

It would be nice if plugin users could at least opt out of requiring background location permission. Perhaps a variable could be set for the plugin that would prevent `ACCESS_BACKGROUND_LOCATION` from being added to the manifest? I would be happy to look in to submitting a PR for this in the coming weeks.

jholzer-cciq, 2021-02-11:
In case it's of any help to anyone, the solution I have gone with for now is to add an `after_prepare` hook script that runs only for Android, and removes the `ACCESS_BACKGROUND_LOCATION` permission declaration from AndroidManifest.xml. This way, I am able to continue using the latest plugin version.

I essentially used the same solution as described in this [StackOverflow post](https://stackoverflow.com/a/45425075/176651), though you can put the script wherever you like within your project's directory tree (does not have to be in root folder).

randdusing, 2021-02-14:
It sounds like I should revert this PR (https://github.com/randdusing/cordova-plugin-bluetoothle/pull/654/files) and the permission should be added manually or via these separate plugins: https://github.com/randdusing/cordova-plugin-background-mode-bluetooth-peripheral/blob/master/plugin.xml or https://github.com/randdusing/cordova-plugin-background-mode-bluetooth-central/blob/master/plugin.xml

What do y'all think? I no longer have an Android device, so I can't test this myself

NunoSav, 2021-02-15:
I'm all for that.

I haven't been able to test it myself yet but looking at other forums/discussions if the plugin is using the permission and the developer has not been able to justify its use then you won't get it published.

jholzer-cciq, 2021-02-15:
I am certainly in favor of reverting the PR, unless someone can explain, or show evidence, how reverting would cause a problem.

I went back and looked at the original issue that was raised (https://github.com/randdusing/cordova-plugin-bluetoothle/issues/650#issue-745176028), and I was unable to reproduce it. I set my target SDK to Android 10 (API Version 29), rebuild and reinstalled the app on my phone (Android 11), and was able to successfully scan for BLE devices in the foreground without the need for background permissions.

Obviously, in the PR author's case, background permissions were needed, but it does not appear to be true in general. Therefore, I think it's sensible to not have background location as the default, especially given the compliance issues it raises when releasing to Google Play.

I've seen no documentation from Google that indicates background location permission is required for scanning in foreground. An overview of BLE permissions from the Android docs can be found [here](https://developer.android.com/guide/topics/connectivity/bluetooth-le#permissions). Only the `ACCESS_FINE_LOCATION` permission is specifically mentioned.

DominikTrenz, 2021-02-15:
I just tested both versions (with and without background location permission) on an android device with targetSdkVersion 30 and it seems to work. So the problem/bug that caused us to add the permission in the first place (sdk 29) seems to be fixed.

TomBrouws, 2021-02-16:
I've also tested our app with both the current and previous versions of the plugin (without the `ACCESS_BACKGROUND_LOCATION` permission), for target SDK 29 and 30. In all cases scanning of BT devices works fine in our use case, which is foreground only.

I'm in favour of reverting the change so we can get back to using the latest version.

randdusing, 2021-02-19:
This has been removed in 6.2.2!

closed 2021-04-22 by NunoSav