randdusing/cordova-plugin-bluetoothle #668: iOS - how to tell if device is already paired

gregor-srdic, 2021-02-04:
I scan for nearby devices and display them to the user. Upon click, the app tries to establish Bluetooth connection. 
Is it possible to tell if device was previously already paired? 
If it is the first time, I need to show the user a dialog with device PIN (which he must enter into the system dialog). If it was already connected before, this is not required and I do not want to show the dialog with the PIN. 
wasConnected operation returns false - so it is not helpful :S

phatpaul, 2021-07-14:
If a device is already paired/bonded, then it won't be discovered with startScan().  (this is my experience on Android 7 ~ 10 and iOS)
You have to also check for bonded devices with retrieveConnected().  

Also, I didn't have to add any logic to my app to show a pairing dialog (PIN entry), that was automatically handled by the OS if the BLE peripheral device required it.

aroncal, 2022-07-05:
I also have a peripheral with PIN security enabled.

On Android, retrieveConnected() returns only bonded devices, whereas on IOS (14.6) returns a connected but still not bonded peripheral (even while IOS is showing the prompt to enter the PIN of the peripheral to make the bonding).

On Android I connect, and while the system prompt for bonging and PIN entering is present, I keep polling for bonded devices every second until the connected one appears; only then I know that the device is connected and bonded, and continue with other operations (writing on characteristics for example).

On IOS, I can't make this polling, because the connected device is returned by 'retrieveConnected()' from the very first instant after connection. Therefore, I don't know how to properly check the moment when the user has entered the correct PIN, to continue with other operations... Also, we don't have the bond() operation on IOS.

Any other ideas for this kind of workflow?

phatpaul, 2022-07-05:
I think you should start a read or write to a characteristic that requires encrypted connection.  That would trigger the pairing dialog if it isn't already completed.  Then wait on the callback from that read/write.  That's what I'm doing anyway.