randdusing/cordova-plugin-bluetoothle #670: on iOS, startScan function always return result "ScanStarted"

parsehari, 2021-02-16:
I am not getting the ScanResult status in success with the StartScan function.

bluetoothle.startScan(startScanSuccess, handleError, {
			"services": [] OR "services": ["180D","180F"],
			"allowDuplicates": true
		});

function startScanSuccess(result) {

	log("startScanSuccess(" + result.status + ")");

	if (result.status === "scanStarted") {

		log("Scanning for devices (will continue to scan until you select a device)...", "status");
	} else if (result.status === "scanResult") {

		if (!foundDevices.some(function (device) {

				return device.address === result.address;

			})) {

			log('FOUND DEVICE:');
			log(result);
			foundDevices.push(result);
			log("device found ", foundDevices);
			//addDevice(result.name, result.address);
		}
	}
}

MojioMS, 2021-02-16:
startScan will not respond with an error if bluetooth is disabled on the device. Instead, the startScan will just not return any results.
Please make sure bluetooth is enabled on your device and try again.

parsehari, 2021-02-17:
Hi, Yes I verified that Bluetooth is enabled. later on my code stuck  at below condition only

 if (result.status === "scanStarted") {
   log("Scanning for devices (will continue to scan until you select a device)...", "status");
}

marutifh, 2021-09-28:
Hi @parsehari, 
Even I have same issue. Did you figure out the solution to this? 

On Android device, I am able to get results of iOS devices, but on Android to Android, iOS to Android no response, service just start scan after that nothing.

@randdusing

filipap, 2021-11-16:
Hi @marutifh @parsehari, after some research I made a workaround solution that solves the scanning problem with Android devices on peripheral mode on iOS and decreases that on Android. 
This week I'll make a merge request with the solution I came up with. 

If you want to change manually the plugin to test, all you have to do is:

1. Change file src/android/BLE/BluetoothLePlugin.java lines 1149 to 1180 and use this:
```java 
      /* build the ScanSetting */
      ScanSettings.Builder scanSettings = new ScanSettings.Builder();
      
      if(obj.length() > 1) {
        scanSettings.setReportDelay(0);
        int scanMode = obj.optInt(keyScanMode, ScanSettings.SCAN_MODE_LOW_LATENCY);
        try {
          scanSettings.setScanMode(scanMode);
        } catch (java.lang.IllegalArgumentException e) {
        }
  
        if (android.os.Build.VERSION.SDK_INT >= android.os.Build.VERSION_CODES.M) {
          int matchMode = obj.optInt(keyMatchMode, ScanSettings.MATCH_MODE_AGGRESSIVE);
          try {
            scanSettings.setMatchMode(matchMode);
          } catch (java.lang.IllegalArgumentException e) {
          }
  
          int matchNum = obj.optInt(keyMatchNum, ScanSettings.MATCH_NUM_MAX_ADVERTISEMENT);
          try {
            scanSettings.setNumOfMatches(matchNum);
          } catch (java.lang.IllegalArgumentException e) {
          }
  
          int callbackType = obj.optInt(keyCallbackType, ScanSettings.CALLBACK_TYPE_ALL_MATCHES);
          try {
            scanSettings.setCallbackType(callbackType);
          } catch (java.lang.IllegalArgumentException e) {
          }
        }
      }
      
      // Start the scan with or without service UUIDs
      if(uuids.length > 0) {
        bluetoothAdapter.getBluetoothLeScanner().startScan(scanFilter, scanSettings.build(), scanCallback);
      }
      else {
        bluetoothAdapter.getBluetoothLeScanner().startScan(scanCallback);
      }
```
Basically what I do here is instead of manually adding default settings, I use the ones instantiated by the ScanSettings.Builder. 

2. On iOS what I do is remove all the options related to the scan mode, 
so I changed file src/ios/BluetoothLePlugin.m line 780 from this:
```objectivec
  [centralManager scanForPeripheralsWithServices:serviceUuids options:@{ CBCentralManagerScanOptionAllowDuplicatesKey:allowDuplicates }];
``` 
to this: 
```objectivec
  [centralManager scanForPeripheralsWithServices:serviceUuids options:nil}];
``` 
Hope it helps 😄 .