randdusing/cordova-plugin-bluetoothle #680: Android 10, no devices detected during scann

oxsmose, 2021-03-15:
Hello,

I deciced to switch on Android device but unfortunatly scann function resturn no devices:
Initialization return "enable"scann return "scanStarted" but no devices found.

I use the following function:
bluetoothle.startScan(startScanSuccess, handleError, { services: [] });

Any idea of what I did wrong ?

If I go in android bluetooth setup the device is detected and I can can pair it.
The device has a passcode. Not sure if it can explain the issue.

ab12gu, 2021-06-24:
Do you have the code example for your startScanSuccess function?

grassick, 2022-02-15:
Here's the one I'm using. Same issue on Android 11 with SGH-I337M:

```
bluetoothle.startScan(function(s) { console.log("success"); console.log(s) }, function(e) { console.error(e) }, { services: []});
```

It doesn't find any devices. But if device is already paired, it works great. Previous calls were:

```
bluetoothle.initialize(function(r) { console.log(r) }, { request: true })
```

which returned `{status: "enabled"}`

grassick, 2022-02-15:
I needed to do `bluetoothle.requestPermission(requestPermissionSuccess, requestPermissionError);` first!