randdusing/cordova-plugin-bluetoothle #681: Unable to implement BluetoothLE in the constructor

kobrecht12, 2021-03-16:
I am attempting to add "private bluetoothLE: BluetoothLE" in my constructor but it is saying that BluetoothSerial is a value and it instead suggests BluetoothLEOriginal. If I add this to the import statement and also change the constructor to be "private bluetoothLE: BluetoothLEOriginal" it works and I can use all of the functions, but when I try to run "ionic cordova run android --device", I get an error saying:
"export 'BluetoothLEOriginal' was not found in '@ionic-native/bluetooth-le'
Does BluetoothSerial still work? If so, how do I properly implement it?

aselley, 2021-03-16:
Hi, for me if I replace:
`import { BluetoothLe } from '@ionic-native/bluetooth-le'`
with:
`import { BluetoothLE } from '@ionic-native/bluetooth-le/ngx';`

it works, hopefully this works for you too.

kobrecht12, 2021-03-16:
> Hi, for me if I replace:
> `import { BluetoothLe } from '@ionic-native/bluetooth-le'`
> with:
> `import { BluetoothLE } from '@ionic-native/bluetooth-le/ngx';`
> (everywhere you import it, i.e: app.module.ts and then any page/service that you inject bluetooth into with the constructor)
> it works for me, hopefully this works for you too.

You're a lifesaver, thanks!