randdusing/cordova-plugin-bluetoothle #682: Readme has conflicting android minimum requirements

ab12gu, 2021-03-28:
"Android 4.3 or higher, Android Cordova library 5.0.0 or higher, target Android API 23 or higher"

Platform: Android 4.3 supports API 18 or lower
Android API 23 supports Android Platform 6.0 or higher

Since API 23 is the higher of the two, I believe the platform should be 6.0 rather than 4.3.
If I am mistaken, please let me know.

randdusing, 2021-03-29:
Mind raising a PR? :)

ab12gu, 2021-03-29:
Yeah I can do that :)

closed 2021-04-15 by ab12gu

alexhisen, 2021-06-24:
Actually, Android Cordova 5.x can be used with much higher target APIs. Those two aspects are independent.

ab12gu, 2021-06-24:
I'm confused about this comment. The updated readme with this PR should address what you stated. See solution: #686

alexhisen, 2021-06-24:
The update is not actually accurate. This plugin DOES work with Android Cordova 5 just fine as long as you target API 23 or higher.

ab12gu, 2021-06-24:
"Android Cordova library 5.0.0 or higher target Android API 23/Platform 6.0 or higher (support for older Android versions should use versions 2.4.0 or below)"

Unsure how this isn't the same thing? Unless you are nitpicking the wording... there was an accidental comma, but thats the only issue I saw...

alexhisen, 2021-06-24:
I guess the Platform 6.0 part is confusing and unnecessary.

ab12gu, 2021-06-24:
Seems standard: https://source.android.com/setup/start/build-numbers shows versioning which is just the specific android platform android x [1 -> 12+]. They use platform terminology: https://developer.android.com/studio/releases/platforms.