randdusing/cordova-plugin-bluetoothle #701: Unable to use with BT scales without pairing and address hacks

EverettMcKay, 2021-07-14:
I have been using bluetoothle for an Android app that connects to a BT scale for about 18 months now. To get it to work, I had to resort to using a couple hacks that are now becoming usability problems. It’s finally time to refresh the code to eliminate these problems and clean things up a bit, but this effort was not successful. I’m hoping you can help me figure out why.

Before getting started: 1) I have upgraded to the latest version of bluetoothle, which didn’t appear to change anything, 2) yes, I have location services enabled, and 3) apps exist that do what I want (BLE Scanner), so I know this should be possible.

Problem 1: Can’t scan certain scales, so have to hardwire their BT address

I’m testing with two scales, startScan works as expected for the Xiaomi, but returns nothing for the Seca (which uses CHIPSEA-BLE). Here is the code I’m using:
```
  const scaleServiceUuid = "181b";
  const params = {
    services: [scaleServiceUuid],
    allowDuplicates: false, // this dont seem to make a difference
    scanMode: bluetoothle.SCAN_MODE_LOW_LATENCY,
    matchMode: bluetoothle.MATCH_MODE_AGGRESSIVE,
    callbackType: bluetoothle.CALLBACK_TYPE_ALL_MATCHES,
  };
  bluetoothle.startScan(btStartScanSuccess, btStartScanError, params);
}
```
And the calls that precede this:
•	initialize
•	requestPermission
•	isLocationEnabled
which are all successful.

My work around is to get the scale’s BT address using BLE Scanner, then have the user enter it into the app. Not very slick.

Problem 2: Must pair/bond

Pairing on the Android device we are using is pretty annoying. To pair, you of course go to Bluetooth settings, but pairing with the CHIPSEA-BLE gets a “Pairing rejected by CHIPSEA-BLE” message but this pairing does make a difference. Worse, I presume because of this rejection, you have to re-pair every time your restart the device or it loses power.

My understanding is that bonding/pairing is optional, and I would prefer not to use them. However, without pairing, startScan returns a “scanStarted” but never a “scanResult”. Pairing fixes this problem.

```
function btStartScanSuccess(obj) {
  if (obj.status === "scanStarted") {
    scanTimer = setTimeout(scanTimeout, btTimeout);
    return;
  }
  if (obj.status === "scanResult") {
    // never received unless the user pairs the device
```
The app would be much better if users could skip this pairing process. However, even if you pair, I find that I must call bond:
```
      const params = { address: scaleAddress };
      bluetoothle.bond(btBondSuccess, bondError, params);
```
connect fails unless bond is called first:
```
  const params = { address: address, autoConnect: true }; 
  bluetoothle.connect(btConnectSuccess, connectError, params);
  connectTimer = setTimeout(connectTimeout, btTimeout);
```
Bonus problem: must call discover

Finally, I have found that subscribe doesn’t work unless discover is called:
```
function btConnectSuccess(obj) {
  if (obj.status === "connected") {
    if (connectTimer) {
      clearTimeout(connectTimer);
      connectTimer = null;
    }
    const params = {
      address: scaleAddress,
      name: scaleName,
      clearCache: false,
    };
    bluetoothle.discover(
      btDiscoverSuccess, (obj) => { dump("discover error", obj, true); },
      params
    );
```
I know the services these devices have, so I shouldn’t have to discover them each time. Not a usability problem, but seems unnecessary and inefficient. I fix would be nice but certainly not essential.

Any help with any of these problems would be greatly appreciated. Since other apps can to connect to this Seca scale and subscribe to the service I need, so I know it’s possible somehow.

randdusing, 2021-07-17:
You may try adjusting or removing the last 3 flags of that startScan params. It's been a while, but I remember certain combinations caused scanning to not work for me.

Unfortunately discovering is required even if you already know the services and characteristics.

EverettMcKay, 2021-07-17:
Thanks, Rand. I will try adjusting the flags.

In the meantime, I just stumbled across a detail that might lead to a solution for this scale. I was messing around with Web Bluetooth and noticed that while this works for most scales:

```
const scaleServiceUuid = 0x181b;
const bluetoothDevice = await navigator.bluetooth.requestDevice({ filters: [{ services: [scaleServiceUuid] }] });
```

This does not work for the Seca scale. But surprisingly this does:

```
const options = {
   acceptAllDevices: true,
   optionalServices: [scaleServiceUuid]
 };
const bluetoothDevice = await navigator.bluetooth.requestDevice(options);
```

So, is there a bluetoothle way to request optionalServices?

You would think both versions would have the same result, but they don't.

EverettMcKay, 2021-07-21:
I finally have everything working in an acceptable manner. The solution: get the scale without asking for scales. So, this works:
```
  const params = {
    services: [],
  };
  bluetoothle.startScan(btStartScanSuccess, btStartScanError, params);
```

but this doesn't:
  ```
const params = {
    services: [scaleServiceUuid],
  };
  bluetoothle.startScan(btStartScanSuccess, btStartScanError, params);

```
This makes no sense to me (and why it took me so long to try). Seems like the obvious step 1 in BT programming is to request devices that support a set of services, and that devices that support that service should respond.

closed 2021-07-21 by EverettMcKay