randdusing/cordova-plugin-bluetoothle #706: Build Error with latest Version

EinfachHans, 2021-07-28:
With the merged PR #696 i can not build my App anymore:

**variable returnObj is already defined in method onNotificationSent(android.bluetooth.BluetoothDevice,int)**

We have `JSONObject returnObj = new JSONObject();` in line 4748 and in line 4728

randdusing, 2021-07-28:
@VinardoZzZ2000 Can you fix this asap? If not, I may need to revert the PR. I don't have any Android devices, so I'm not confident in making the changes myself

phatpaul, 2021-07-28:
I'm also getting the build error.  I reverted this plugin to 6.5.1 for now.

EinfachHans, 2021-07-28:
Also i don't like the fact that it could cause a callback to be fired twice^^

tigohenryschultz, 2021-07-28:
[image]

randdusing, 2021-07-30:
I reverted this

VinardoZzZ2000, 2021-07-31:
Very sorry about this. My project uses a build without the old callback, so I thought there wouldn't be any problems just adding back the old callback, and skipped the build process. I'll prepare another PR soon. Would it be preferrable to also remove the old callback immediately instead of deprecating it?

EinfachHans, 2021-08-05:
As this is reverted, i will close this issue

closed 2021-08-05 by EinfachHans