randdusing/cordova-plugin-bluetoothle #711: iOS read characteristic, w 128 service UUID returns service not found

sdetweil, 2021-08-23:
I did a scan, found peripherals, connected, now want to read a characteristic

using the advertised service UUID returned win the peripheral, which in my case is a private/custom UUID.

all the samples show the 4 digit UUID.

sdetweil, 2021-08-23:
log from my app scan to read characteristic
```
 [log] - BT initialize=[object Object]
⚡️  [log] - start scanning
⚡️  [log] - scan for BluetoothLE
To Native Cordova ->  BluetoothLePlugin initializePeripheral BluetoothLePlugin907918909 ["options": [<null>]]
2021-08-22 13:24:36.002034-0500 App[1364:656222] [CoreBluetooth] WARNING: <CBPeripheralManager: 0x282dfe580> has no restore identifier but the delegate implements the peripheralManager:willRestoreState: method. Restoring will not be supported
To Native Cordova ->  BluetoothLePlugin startScan BluetoothLePlugin907918910 ["options": [{
}]]
⚡️  [log] - Scanning for devices (will continue to scan until you stop it)... status
⚡️  [log] - FOUND DEVICE:
⚡️  [log] - {
  "status": "scanResult",
  "advertisement": {
    "solicitedServiceUuids": [],
    "overflowServiceUuids": [],
    "localName": "SamsDevice",
    "isConnectable": 1,
    "serviceData": {},
    "serviceUuids": [
      "00020202-27B9-42F0-82AA-2E951747BBF9"
    ]
  },
  "rssi": -82,
  "name": "SamsDevice",
  "address": "9DC84DDB-DCB5-8B9D-FAD3-D76C56DB31BD"
}
⚡️  [log] - FOUND DEVICE:
⚡️  [log] - {
  "status": "scanResult",
  "advertisement": {
    "solicitedServiceUuids": [],
    "overflowServiceUuids": [],
    "localName": "SamsDevice",
    "isConnectable": 1,
    "serviceData": {},
    "serviceUuids": [
      "00010202-27B9-42F0-82AA-2E951747BBF9"
    ]
  },
  "rssi": -50,
  "name": "SamsDevice",
  "address": "FF85CCC9-5925-DFB4-C2DE-A614F966D654"
}
To Native Cordova ->  BluetoothLePlugin stopScan BluetoothLePlugin907918913 ["options": []]
⚡️  [log] - stopscan successful
⚡️  [log] - processinging devices
To Native Cordova ->  BluetoothLePlugin connect BluetoothLePlugin907918914 ["options": [{
    address = "9DC84DDB-DCB5-8B9D-FAD3-D76C56DB31BD";
}]]
To Native Cordova ->  BluetoothLePlugin connect BluetoothLePlugin907918915 ["options": [{
    address = "FF85CCC9-5925-DFB4-C2DE-A614F966D654";
}]]
⚡️  [log] - connected, reading rssi from FF85CCC9-5925-DFB4-C2DE-A614F966D654
To Native Cordova ->  BluetoothLePlugin descriptors BluetoothLePlugin907918916 ["options": [{
    address = "FF85CCC9-5925-DFB4-C2DE-A614F966D654";
    characteristic = 9AFD;
    descriptor = 2902;
    service = "00010202-27B9-42F0-82AA-2E951747BBF9";
}]]
⚡️  [log] - read table error={"message":"Service not found","name":"SamsDevice","error":"service","address":"FF85CCC9-5925-DFB4-C2DE-A614F966D654"}
To Native Cordova ->  BluetoothLePlugin descriptors BluetoothLePlugin907918917 ["options": [{
    address = "9DC84DDB-DCB5-8B9D-FAD3-D76C56DB31BD";
    characteristic = 9AFD;
    descriptor = 2902;
    service = "00020202-27B9-42F0-82AA-2E951747BBF9";
}]]
⚡️  [log] - connected, reading rssi from 9DC84DDB-DCB5-8B9D-FAD3-D76C56DB31BD
⚡️  [log] - read table error={"message":"Service not found","name":"SamsDevice","error":"service","address":"9DC84DDB-DCB5-8B9D-FAD3-D76C56DB31BD"}
```

and the code... scan puts the devices in an array, and then I process the array
```
 foundDevices.forEach((peripheral)=>{
              BluetoothLE.connect({address: peripheral.address}).subscribe((device)=>{
                console.log("connected, reading rssi from "+device.address)         
                BluetoothLE.read(
                  {
                    "address": device.address,
                    "service": peripheral.advertisement.serviceUuids[0],
                    "characteristic": "9AFD",
                    "descriptor": "2902"                    
                  }
                ).then((appRssi)=>{
```
and attached is a screen shot from BLE scanner
[image]

sdetweil, 2021-08-24:
well, it turns out that you have to do a services() and characteristics() request, to get the library data filled in. then you can do a read or write..

why doesn't the library do this automatically on connect()?

none of the examples do this...

sdetweil, 2021-08-25:
closing. not a library problem

closed 2021-08-25 by sdetweil

upupzealot, 2021-12-31:
> well, it turns out that you have to do a services() and characteristics() request, to get the library data filled in. then you can do a read or write..
> 
> why doesn't the library do this automatically on connect()?
> 
> none of the examples do this...

very agree, `connect` should do this automatically!

kieranbarlow, 2022-08-08:
This helped me!