randdusing/cordova-plugin-bluetoothle #715: performance questions

sdetweil, 2021-10-11:
My App scans then needs to read characteristics from each detected device with my UUID. 
for 6 devices (started overlapping) this takes 6.5 seconds..  seems a lot of that time is in discover, which appears to be required for each device after connect.  

is there any capability to reuse the received discover info to avoid the performance impact of this step? my devices will not change characteristics from the last read. 
or any other suggestion
I will post ms timings when I am booted back to linux
```
scan
```
then loop for discovered(scanned) devices (submitted in parallel)
```
   connect
       discover 
           read characteristic
           (maybe 2, submitted in parallel)
    disconnect
    close
```               

this is Ionic/Vue on IOS (15.1)

NunoSav, 2021-12-27:
What's the advertising interval you're using on those devices? You have some high values for `connect` and this could help you reduce the average time your cycle takes to run.

sdetweil, 2021-12-27:
@NunoSav  the devices are advertising at 100ms interval.

NunoSav, 2021-12-28:
100ms is short enough. I can't see how you can improve your cycle time any other way, if you find a way to do so please share it :)

sdetweil, 2021-12-28:
yeh, I think there is a problem in IOS locationservices with concurrent connects. 

I haven't tried to build this on android.