randdusing/cordova-plugin-bluetoothle #718: Android 12 Bluetooth Permissions

mosor, 2021-11-17:
This plugin should be updated to work with Android 12's [new bluetooth permissions](https://developer.android.com/guide/topics/connectivity/bluetooth/permissions).

JesperBalslev, 2022-01-24:
Seems we have to continue this plugin ourselves - or change to Capacitor.
Anyone have any experience with Capacitor and their BLE plugin?
Or know of a more updated fork to this plugin?

randdusing, 2022-01-24:
Unfortunately I don't have the time or devices to continue working on this plugin directly, but I'm open to reviewing and hopefully approving pull requests.

JesperBalslev, 2022-01-24:
Hey its ok @randdusing. You have done a great job with this plugin. Thanks.

mosor, 2022-01-24:
I have worked on adding the new runtime permissions, and it does correctly ask for and check for them, but there seems to be issues on Android’s side with pairing peripherals. In the call to connectGatt, there is something logged about not being able to resolve the random private address, which is supposed to be resolved by a key that’s shared during pairing. I’ve tried many different things to try and resolve this issue and there are other people who built custom ble devices that  continue not to work with Android 12 phones.

mosor, 2022-03-18:
Hey @JesperBalslev and @randdusing, I wanted to give a positive update to this issue. I think updating the plugin to request and check for the new permissions is possible. The issue we had is fixable with a firmware change, so people may want to look at their firmware if they still have issues. I'm happy to help when I have free time since I know @randdusing is busy.

Funkelfetisch, 2022-03-25:
@mosor could you please write a PR or send your patch to me? :)
veit@nebulite.berlin

several customers can't scan for their clothing items right now xD

thank you in advance :) <3

Cyb0org, 2022-07-28:
Hi.

I've made PR #730 

I use it currently in one project and it's working fine.

EngageCT, 2024-08-21:
Hi Guys does this  plugin support Android 14? I have tried to test on Android 14 broadcasting but not connecting.  If l target API 33 t is working. Your assistance will be much appreciated. Ionic Cordova

Sebartus, 2024-11-14:
@EngageCT  In API 34 there is a bug in this plugin and in Android settings you have to find your app and add manualy permission for devices nearby to: allow then everything works fine.

Could anybody help with update for this: @Cyb0org  @mosor

Cyb0org, 2024-11-15:
I'm not using Cordova now. This ticket is about A12, so it should be closed and new one about A14 should be created.