randdusing/cordova-plugin-bluetoothle #727: Peripheral not visible

treb93, 2022-06-06:
Hello, 

I recently tried to implement the peripheral cycle on my ionic / angular app, without success.
Previously I used https://github.com/don/cordova-plugin-ble-peripheral, which is working well but is not maintained anymore so that I needed to switch.

Here is the code, I used the `@awesome-cordova-plugin/bluetoothle` adapter for having all the pipe in one block  :
(the result is the same using the cordova plugin directly, which I can submit but is far longer)
``
    from(this.ble.requestPermission())
      .pipe(
        // Initialize bluetooth
        switchMap(() => this.ble.initialize()),

        // Initialize peripheral
        switchMap((response) =>
          this.ble.initializePeripheral({
            request: true,
            restoreKey: 'bluetoothleplugin',
          })
        ),

        // Add service on enabling
        filter((initializeResult) => initializeResult.status === 'enabled'),
        switchMap(() =>
          this.ble.addService({
            service: '1234',
            characteristics: [
              {
                uuid: 'ABCD',
                permissions: {
                  read: true,
                  write: true,
                },
                properties: {
                  read: true,
                  writeWithoutResponse: true,
                  write: true,
                  notify: true,
                  indicate: true,
                },
              },
            ],
          })
        ),

        // Start advertising
        switchMap(() =>
          from(
            this.ble.startAdvertising({
              service: '1234', //Android
              mode: 'lowLatency',
              connectable: true,
              includeDeviceName: true,
              includeTxPowerLevel: true,
            })
          )
        )
      )
      .subscribe((status) => {
        console.log('Advertising status', status);
      });
``

No errors are thrown (also in Android studio logcat), and the `startAdvertising` return seems good : 
``
{
    "mode": 2,
    "timeout": 1000,
    "txPowerLevel": 2,
    "isConnectable": true,
    "status": "advertisingStarted"
}
``

Do I miss something ?

marcui13, 2023-02-03:
Hi, 
Were you able to solve this problem? 
I'm having the same issue and I can't solve it

treb93, 2023-02-03:
Nope ! 
Any support is still welcome ^^

jcconneqtech, 2023-04-05:
This works for me: 

```
this.ble.startAdvertising(
    (success) => {
        console.log(success);
    },
    (error) => {
        console.log(error);
    },
    {
        services: ['FE65'], //iOS
        service: 'FE65', //Android
        name: 'advertisingname',
        timeout: 0,
    },
);
```

marcelinojung, 2023-12-25:
timeout: 0

bootexe, 2024-09-05:
> This works for me:
>
> ```
> this.ble.startAdvertising(
>     (success) => {
>         console.log(success);
>     },
>     (error) => {
>         console.log(error);
>     },
>     {
>         services: ['FE65'], [...]

Hello can someone explain me this ? iam lost as f**k
Iam using anddusing/cordova-plugin-bluetoothle on capacitor and the initializePeripheral function seems to work but nothing happens and the status stays has not initialized
iam sure its a stupid error but I cant figure it out
Help me pls

Here is my code

```

      const ble = require('@awesome-cordova-plugins/bluetooth-le')
      const bluetoothle = ble.BluetoothLE

       bluetoothle.initializePeripheral((status)=>{

            console.log('intible',status);
            bluetoothle.addService((...d)=>{
                console.log(d);
                console.log(bluetoothle.hasPermissionBtAdvertise(console.log));
                bluetoothle.startAdvertising(console.log,console.err,{
                    services:["181C"],
                    service:"181C",
                    name:"republica_im" 
                });
            },
            console.err,
            {
                service: "181C",
                characteristics: [
                  {
                    uuid: "181C",
                    permissions: {
                      read: true,
                      write: true,
                      //readEncryptionRequired: true,
                      //writeEncryptionRequired: true,
                    },
                    properties : {
                      read: true,
                      writeWithoutResponse: true,
                      write: true,
                      notify: true,
                      indicate: true,
                      //authenticatedSignedWrites: true,
                      //notifyEncryptionRequired: true,
                      //indicateEncryptionRequired: true,
                    }
                  }
                ]
            });
        },
        console.err,
        {
            "request": true,
            "restoreKey": "_im"
        });

```