randdusing/cordova-plugin-bluetoothle #732: Attempt to invoke virtual method 'boolean java.util.LinkedList.add(java.lang.Object)' on a null object reference

markokitanoski1, 2022-08-10:
Hello there,

I am trying to implement Bluetooth connection to external Bluetooth device, but I am experiencing several issues that I can not solve:
- documentation prefers avoiding the .reconnect() method with using combination of .close & .connect methods instead, but after successfully connecting to a device along with .connect method, device just disconnects everytime in a matter of several (approx. 6s) seconds, and causing a connect - disconnect loop.
- I have implemented a callback which executes whenever device is disconnected to call .reconnect method but I suspect or I discovered that the .read method which I am trying to call when the device is successfully connected throws an error: _Attempt to invoke virtual method 'boolean java.util.LinkedList.add(java.lang.Object)' on a null object reference_ when is beeing called after the .reconnect method. If I do not use reconnect method and use above implementation it causes the connect - disconnect loop but meanwhile .read function executes successfully.

I do not found any workaround.

markokitanoski1, 2022-08-24:
Can this be urgently fixed by the dev team because my functionality which is using this plugin is the core of my application.

Thanks in advance!

davidofwatkins, 2022-12-07:
Any updates on progress or workarounds for this?

I recently upgraded from `v4.4.3` to `v6.7.0` and started receiving this error after a BLE device disconnects unexpectedly and then reconnects. I was able to do a `git bisect` on this plugin within my app and determined that https://github.com/randdusing/cordova-plugin-bluetoothle/pull/694 introduced the broken change.

I'll continue to investigate the issue, but maybe @ttongsul has some ideas?

ttongsul, 2022-12-30:
Hello @davidofwatkins and others that may have been affected by this issue. I'm deeply sorry for the problems that my commit may have caused.

It seems that the cause is in this line [here](https://github.com/randdusing/cordova-plugin-bluetoothle/blob/0f05d77bff65a1b43eff18b4d250cf0abb49a25d/src/android/BluetoothLePlugin.java#L3308), where `connection.get(keyQueue)` may return null instead of a linked list. A possible solution is adding the code `if (queue == null) {
  queue = new LinkedList<Operation>();
  connection.put(keyQueue, queue);
}` next to the line.

Unfortunately, I don't have a working Android dev environment set up and my hands are full at the moment. It would be greatly appreciated if someone could test the solution and open up a PR.

qpwr, 2023-01-09:
I can confirm @ttongsul s patch is working fine for me.

DonOzOn, 2023-02-03:
i added below code to reconnectAction method in Native code and it worked.
    LinkedList<Operation> queue = new LinkedList<Operation>();
    connection.put(keyQueue, queue);
    
[image]