randdusing/cordova-plugin-bluetoothle #733: Nothing happens on all methods

matirodriguez, 2022-08-24:
Im using

    "@awesome-cordova-plugins/bluetooth-le": "^5.44.0",
    "cordova-plugin-bluetoothle": "^6.7.0",
    "@ionic/angular": "^6.1.9",

Whenever I call a method (requestPermissions, initialize, getAdapterInfo, startScan) nothing happens,
It doesnt go neither to the success nor the error, Also no console errors, also no Android errors,

Did anyone experienced this? Thanks

draudrau, 2022-09-09:
For me it's almost the same thing.
sequence of plugin calls :
- isInitialized -> { isInitialized: false }
- initialize -> { status: 'enabled' }
- hasPermission -> { hasPermission: false }
- requestPermission -> { requestPermission: false }
Android doesn't prompt for permission

draudrau, 2022-09-09:
My bad !
New methods to use : hasPermissionBtScan and requestPermissionBtScan
It works well :-)

Nickholas, 2022-09-10:
I cant see that new methods despite i have installed the 6.7.0 version of cordova-plugin-bluetoothle:

`Property 'hasPermissionBtScan' does not exist on type 'BluetoothLE'`

Any ideas about what i am doing wrong?

Nickholas, 2022-09-10:
> 

Well, i have found that i have the last version of the plugin called "cordova-plugin-bluetoothle", BUT it is NOT updated on the package "@awesome-cordova-plugins/bluetooth-le". So, i cant use that functions on my capacitor proyect.

Is there another way to update it?

Thanks in advance

kabaehr, 2022-10-06:
Hey, I have the same issue with using "@awesome-cordova-plugins/bluetooth-le" there is a PR already, but it is not merged yet. https://github.com/danielsogl/awesome-cordova-plugins/pull/4320 I also mention my workaround there