randdusing/cordova-plugin-bluetoothle #734: Cannot build on graddle 8.0

sidr555, 2022-09-20:
I have errors when trying to build with cordova 10, graddle 8.0, AndroidSDK 30 or 32

> Task :app:compileDebugJavaWithJavac FAILED
/home/sidr/dev/exo/exotrack/platforms/android/app/src/main/java/com/randdusing/bluetoothle/BluetoothLePlugin.java:929: error: cannot find symbol
    addProperty(returnObj, "hasPermission", cordova.hasPermission(Manifest.permission.BLUETOOTH_SCAN));
                                                                                     ^
  symbol:   variable BLUETOOTH_SCAN                                                           
  location: class permission
/home/sidr/dev/exo/exotrack/platforms/android/app/src/main/java/com/randdusing/bluetoothle/BluetoothLePlugin.java:944: error: cannot find symbol
    cordova.requestPermission(this, REQUEST_BLUETOOTH_SCAN, Manifest.permission.BLUETOOTH_SCAN);                                                    
                                                                               ^
  symbol:   variable BLUETOOTH_SCAN       
  location: class permission
/home/sidr/dev/exo/exotrack/platforms/android/app/src/main/java/com/randdusing/bluetoothle/BluetoothLePlugin.java:954: error: cannot find symbol
    addProperty(returnObj, "hasPermission", cordova.hasPermission(Manifest.permission.BLUETOOTH_CONNECT));
                                                                                     ^                                                                                                       
  symbol:   variable BLUETOOTH_CONNECT
  location: class permission                                                                                                                                                                 
/home/sidr/dev/exo/exotrack/platforms/android/app/src/main/java/com/randdusing/bluetoothle/BluetoothLePlugin.java:969: error: cannot find symbol
    cordova.requestPermission(this, REQUEST_BLUETOOTH_CONNECT, Manifest.permission.BLUETOOTH_CONNECT);
                                                                                  ^
  symbol:   variable BLUETOOTH_CONNECT                                                                                                                                                       
  location: class permission                                                                                                                                                                 
/home/sidr/dev/exo/exotrack/platforms/android/app/src/main/java/com/randdusing/bluetoothle/BluetoothLePlugin.java:979: error: cannot find symbol
    addProperty(returnObj, "hasPermission", cordova.hasPermission(Manifest.permission.BLUETOOTH_ADVERTISE));
                                                                                     ^
  symbol:   variable BLUETOOTH_ADVERTISE
  location: class permission
/home/sidr/dev/exo/exotrack/platforms/android/app/src/main/java/com/randdusing/bluetoothle/BluetoothLePlugin.java:994: error: cannot find symbol
    cordova.requestPermission(this, REQUEST_BLUETOOTH_ADVERTISE, Manifest.permission.BLUETOOTH_ADVERTISE);
                                                                                    ^
  symbol:   variable BLUETOOTH_ADVERTISE
  location: class permission
/home/sidr/dev/exo/exotrack/platforms/android/app/src/main/java/com/randdusing/bluetoothle/BluetoothLePlugin.java:1010: error: cannot find symbol
        addProperty(returnObj, "requestPermission", cordova.hasPermission(Manifest.permission.BLUETOOTH_SCAN));
                                                                                             ^
  symbol:   variable BLUETOOTH_SCAN
  location: class permission
/home/sidr/dev/exo/exotrack/platforms/android/app/src/main/java/com/randdusing/bluetoothle/BluetoothLePlugin.java:1013: error: cannot find symbol
        addProperty(returnObj, "requestPermission", cordova.hasPermission(Manifest.permission.BLUETOOTH_CONNECT));
                                                                                             ^
  symbol:   variable BLUETOOTH_CONNECT
  location: class permission
/home/sidr/dev/exo/exotrack/platforms/android/app/src/main/java/com/randdusing/bluetoothle/BluetoothLePlugin.java:1016: error: cannot find symbol
        addProperty(returnObj, "requestPermission", cordova.hasPermission(Manifest.permission.BLUETOOTH_ADVERTISE));
                                                                                             ^
  symbol:   variable BLUETOOTH_ADVERTISE
  location: class permission
Note: Some input files use or override a deprecated API.
Note: Recompile with -Xlint:deprecation for details.
Note: Some input files use unchecked or unsafe operations.
Note: Recompile with -Xlint:unchecked for details.
9 errors

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':app:compileDebugJavaWithJavac'.
> Compilation failed; see the compiler error output for details.

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output. Run with --scan to get full insights.

* Get more help at https://help.gradle.org

Deprecated Gradle features were used in this build, making it incompatible with Gradle 8.0.

byblik, 2022-09-22:
@sidr555 I stumbled on the same issue and figured it out for myself having installed version 6.6.2.

kabaehr, 2022-10-06:
Hey, we had the same. 
For targetSdk=32, you need to request additional permissions. BLUETOOTH_ADVERTISE and BLUETOOTH_CONNECT, those got added to the this plugin within version 6.70 

You when targeting SDK 32 you need to use plugin version 6.7.0 as well and implement the needed and documented changes as well

northkode, 2022-11-02:
Still having this issue.. and the comments above have not helped? Is there a procedure that I can follow?

jammind, 2022-11-09:
The missing symbols (constants) have been available since Android 12 / API level 31 as documented https://developer.android.com/guide/topics/connectivity/bluetooth/permissions.

I have overridden the targetSdkVersion to 31 in config.xml and it works.

```
    <platform name="android">
        <preference name="android-targetSdkVersion" value="31" />
    </platform>
```