randdusing/cordova-plugin-bluetoothle #736: Android 12 requestPermissionBtConnect - it allows nearby devices BUT crash and exit

Sebartus, 2022-09-22:
In Android 12 ( in Android 11 without this new method everything works fine ) I add new method:
`bluetoothle.requestPermissionBtConnect(requestPermissionSuccess, requestPermissionError);`

In app I got new popup to allow Nearby devices and in Settings >> Location >> App Name I can see it:

[image]

but then app crash and exit. In log i can see error:
`java.lang.SecurityException: Need android.permission.BLUETOOTH_CONNECT permission for AttributionSource`

The only solution is to go Settings >> Location >> App Name and manualy Don’t allow
and again Allow Nearby devices to work correct. This situation occurs both in the simulator and on the real device.

Cyb0org, 2022-09-24:
Do you have `BLUETOOTH_CONNECT` permissions in the `AndroidManifest.xml`?

Sebartus, 2022-09-29:
> Do you have `BLUETOOTH_CONNECT` permissions in the `AndroidManifest.xml`?

Yes I have `BLUETOOTH_CONNECT` but I solved the problem. 

SOLUTION:
Correct sequence is:
1. `bluetoothle.initializePeripheral(success, error, params);`
2. `bluetoothle.hasPermissionBtConnect(hasPermissionSuccess);`
3. `bluetoothle.requestPermissionBtConnect(requestPermissionSuccess, requestPermissionError);` - the lack of this step caused an error
4. bluetoothle.isLocationEnabled(isLocationEnabledSuccess, isLocationEnabledError);
...

closed 2022-09-29 by Sebartus