randdusing/cordova-plugin-bluetoothle #737: Unable to read from Device

ari-sarkar, 2022-10-26:
[image]

I am trying to read from a Flic button but getting the error "Unable to read" while trying to read().

The process is as follows:
1. Calling the scan() to discover devices.
2. Connect() to the device using UUID.
3. Checking for Bluetooth permission and moving to the next step accordingly.
4. Calling discover() to check device info.
5. In this step attaching a subscription to a service using subscribe().
6. **Inside the response of subscribe() calling read() to read button press with correct service, characteristic & UUID.**

error: read,
message: unable to read

marcui13, 2023-02-08:
Hi, how are you?
I have the same problem, were you able to solve that?

ari-sarkar, 2023-02-11:
no