randdusing/cordova-plugin-bluetoothle #738: Connect 2 Android devices using BluetoothLE

bevinGithub, 2022-11-08:
Is it possible connecting 2 android phones with Bluetooth and share/send data to another Android phone 
and sends back a response after receiving and reading the data.

The idea is to send data/share data offline using Bluetooth on Android to Android Phone/tablet

Below is my environment
Ionic:
   Ionic CLI                     : 6.20.1 
   Ionic Framework               : @ionic/angular 6.2.2
   @angular-devkit/build-angular : 13.0.4
   @angular-devkit/schematics    : 13.0.4
   @angular/cli                  : 13.0.4
   @ionic/angular-toolkit        : 5.0.3

Cordova:
   Cordova CLI       : 11.0.0
   Cordova Platforms : android 10.1.2
   Cordova Plugins   : cordova-plugin-ionic-keyboard 2.2.0, cordova-plugin-ionic-webview 5.0.0, (and 14 other plugins)

Utility:
   cordova-res                          : 0.15.4
   native-run (update available: 1.7.1) : 1.5.0

System:
   NodeJS : v14.15.4 (C:\Program Files\nodejs\node.exe)
   npm    : 6.14.10
   OS     : Windows 10

marcui13, 2023-01-31:
Hi how are you?
I want to develop an application with the same functionality that you describe, I would like to know if you were able to do it, please!

bevinGithub, 2023-02-01:
Hi Marcui.

Yes l did manage to get it going with the Bluetooth LE. You need to make the other devices a peripheral and the once that scans the central.

marcui13, 2023-02-01:
Great, I'm glad you made it!

Did you develop the same app for both devices?
Or did you develop two different apps, one for the peripheral and one for the central one?
Did you use this plugin?

Thank you for your answer and sorry for the inconvenience!

bevinGithub, 2023-02-02:
Hi in my case i created a tab for Broadcast and the other for Scan on one app as l need to scan RFID chip with NFC and sends the data to the other device. Some of the chips are high and we needed to be able to scan the chip one guys scans and the other receives the info.