randdusing/cordova-plugin-bluetoothle #739: Crash of the App when StartScan() is called with BLUETOOTH_SCAN permission

MSMTjd, 2022-11-08:
Hi !

My app crash when `startScan()` function is called.

This only happens when the BLUETOOTH_SCAN permission is allowed.
I use the `requestPermissionBtScan()` function, allowed it when the window appear, and when I hit my search button that call `startScan()`, my app is closed without any error message.
I try to use the "android-permission" plugin, and same problems.

Issue #736 seem to be the same problems, but the solution doesn't work for me.

I precise that I'm not an expert and work on a legacy code.

Tweakie, 2023-02-01:
Where you able to solve the issue? I'm experiencing the same issue.

Tweakie, 2023-02-01:
Solved the issue. When you are scanning and you are specifying services to filter on you also need to have Connect permissions. If you don't request them the app crashes on start-up.