randdusing/cordova-plugin-bluetoothle #765: Bluetooth LE not working on Android 14 but working on Android 13

EngageCT, 2024-07-29:
Hi guys l have a problem with Bluetooth LE using the Advertising on ionic Cordova project. The plugin is working fine on Android 13 if l upgrade to Android 14 i.e. API 34 it is not working on Advertising. Is there anyone having the same issue or has a solution for this. My environment is below:

Bluetooth LE is on version 6.4 required 6.8  but l haven't updated the plugin

**Ionic:**

   Ionic CLI                     : 7.2.0 
   Ionic Framework               : @ionic/angular 6.7.5
   @angular-devkit/build-angular : 13.0.4
   @angular-devkit/schematics    : 13.0.4
   @angular/cli                  : 13.0.4
   @ionic/angular-toolkit        : 5.0.3

Cordova:

   Cordova CLI       : 12.0.0 (cordova-lib@12.0.1)
   Cordova Platforms : android 13.0.0
   Cordova Plugins   : cordova-plugin-ionic-keyboard 2.2.0, cordova-plugin-ionic-webview 5.0.0, (and 11 other plugins)

**Utility:**

   cordova-res                          : 0.15.4
   native-run (update available: 2.0.1) : 1.7.2

**System:**

   Android SDK Tools : 26.1.1 
   NodeJS            : v20.15.1
   npm               : 10.8.1
   OS                : Windows 10

EngageCT, 2024-08-06:
Is there anyone with information on Bluetooth LE on android 14 on Ionic cordova

Sebartus, 2024-11-14:
@EngageCT  In API 34 there is a bug in this plugin and in Android settings you have to find your app and add manualy permission for devices nearby to: allow then everything works.

randdusing, 2024-11-14:
I don't have the time or devices to make the changes myself, but I'd be happy to approve a PR.

asif166037, 2025-10-23:
Hi @EngageCT ,  I have tried and working fine in Android API -34 ,  can you share the error.